tsandmann/platform-teensy #10: Standard functions like `snprintf` don't work

arandomliz, 2021-11-22:
I tried formatting a string using `snprintf`. When I do that with this platform instead of the default `teensy` platform, it won't work. The default platform (`teensy`) does. Example code I used:

```cpp
#include <Arduino.h>

void setup() {
  Serial.begin(9600);
}

void loop() {
  float number = 0.2;
  char command[24] = {0};

  size_t n = snprintf(command, sizeof(command), "%f", number);

  Serial.printf("%s | %d\n", command, n);

  delay(1000);
}
```

tsandmann, 2021-11-22:
This platform uses newlib nano by default (see [documentation](https://github.com/bminor/newlib/blob/a92d69d7439de92974d733d402d34f6f11dafdcb/newlib/README#L393)), therefore floating point support for printf / scanf must be enabled explicitly by linking with `-u _printf_float` or adding `asm(".global _printf_float");` somewhere in your code.

You can also add `build_unflags = --specs=nano.specs` to platformio.ini to disable the nano libc version completely.

arandomliz, 2021-11-22:
That fixed the issue. I'm seriously impressed.

Thank you for the amazing support.

closed 2021-11-22 by arandomliz

tsandmann, 2021-11-22:
You're welcome 🙂