tsandmann/platform-teensy #11: Can't add core arduino libraries

krav4, 2023-02-02:
When I am using this version of the platform, I can no longer add library dependencies for arduino core.

Here is a working platformio.ini:
`
; PlatformIO Project Configuration File

[env:teensy41]
platform = teensy
framework = arduino
board = teensy41
lib_deps = 
    SPI
    https://github.com/ftrias/TeensyDebug.git
`

Here is a platformio.ini that generates the error:
`
; PlatformIO Project Configuration File

[env:teensy41]
platform = https://github.com/tsandmann/platform-teensy.git#b9da34f061503ffacc1b771522206f579cbb6a19
framework = arduino
board = teensy41
lib_deps = 
    SPI
    ; NXP-FreeRTOS
    https://github.com/ftrias/TeensyDebug.git
    https://github.com/tsandmann/freertos-teensy.git#72c344205a03fd3a17391b6b6a4328be043172ba
`

Here is the error:
`
Processing teensy41 (platform: https://github.com/tsandmann/platform-teensy.git; framework: arduino; board: teensy41)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CC flags: -DUSB_DUAL_SERIAL -Wno-unused-parameter
CXXFLAGS: -Wno-volatile
PlatformIO running on linux_x86_64
CONFIGURATION: https://docs.platformio.org/page/boards/teensy-ts/teensy41.html
PLATFORM: Teensy (4.17.0+sha.2bd78d3) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (custom) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.157.2+sha.7b0c6da 
 - tool-teensy @ 1.157.0 (1.57) 
 - toolchain-arm-cortexm-linux @ 12.2.0-3+sha.7bc67ae
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_x86_64'
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TeensyDebug @ 0.0.2+sha.d02f575
|-- freertos-teensy @ 10.5.1-4+sha.72c3442
|   |-- NXP-FreeRTOS @ 10.4.3
Building in debug mode
Compiling .pio/build/teensy41/lib8d7/freertos/croutine.c.o
Compiling .pio/build/teensy41/lib8d7/freertos/event_groups.c.o
Compiling .pio/build/teensy41/lib8d7/freertos/list.c.o
Compiling .pio/build/teensy41/lib8d7/freertos/portable/GCC/ARM_CM4F/port.c.o
Compiling .pio/build/teensy41/lib8d7/freertos/queue.c.o
Compiling .pio/build/teensy41/lib8d7/freertos/stream_buffer.c.o
Compiling .pio/build/teensy41/lib8d7/freertos/tasks.c.o
Compiling .pio/build/teensy41/lib8d7/freertos/timers.c.o
Compiling .pio/build/teensy41/src/Blink.cpp.o
Compiling .pio/build/teensy41/src/freertos_hooks.c.o
Compiling .pio/build/teensy41/src/spi.cpp.o
src/Blink.cpp:8:10: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
*
*************************************************************

    8 | #include "SPI.h"
      |          ^~~~~~~
compilation terminated.
*** [.pio/build/teensy41/src/Blink.cpp.o] Error 1
src/spi.cpp:2:10: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
*
*************************************************************

    2 | #include "SPI.h"
      |          ^~~~~~~
compilation terminated.
*** [.pio/build/teensy41/src/spi.cpp.o] Error 1
=========================================================================== [FAILED] Took 0.69 seconds ===========================================================================

 *  The terminal process "platformio 'run', '--environment', 'teensy41'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
`

tsandmann, 2023-02-03:
Thanks for reporting this bug.
I've created an experimental bugfix. Maybe you could test it? Therefore please adjust `platformio.ini` as follows:
```ini
platform = https://github.com/tsandmann/platform-teensy.git#bugfixes/fw_libs
```

You might have to delete the folders `$HOME/.platformio/platforms/teensy-ts` and `$HOME/.platformio/framework-arduinoteensy` because of some name changes.

krav4, 2023-02-03:
Thanks! It worked fine! There are a few warnings during compilation but i am guessing this is because you updated the compiler:
```
/home/vagrant/.platformio/packages/framework-arduinoteensy-ts/teensy4/HardwareSerial.cpp: In member function 'void HardwareSerial::begin(uint32_t, uint16_t)':
/home/vagrant/.platformio/packages/framework-arduinoteensy-ts/teensy4/HardwareSerial.cpp:177:18: warning: unused variable 'tx_fifo_size' [-Wunused-variable]
  177 |         uint16_t tx_fifo_size = 4;
      |                  ^~~~~~~~~~~~
/home/vagrant/.platformio/packages/framework-arduinoteensy-ts/teensy4/HardwareSerial.cpp:179:18: warning: unused variable 'rx_fifo_size' [-Wunused-variable]
  179 |         uint16_t rx_fifo_size = 4;
```

So what was wrong with it? I am using platformio for professional use in a project (not by choice) and would like to understand the problem. I see you updated the python arduino.py script to point to framework-arduinoteensy-ts, but I am not sure what that does. Things are a bit more obfuscated than the good old cmake =)

tsandmann, 2023-02-03:
Thanks for testing!

The problem was that the custom platform referenced a custom teensy cores library, which doesn't have all the standard libraries included (of course). Therefore also the standard core framework has to be installed, so that platformio can find the libraries like SPI there.

The warnings are because of a recent update of the core library which has this unused variables in it - I don't know why. Will update that, too.

commit d16a54c3 (2023-02-03): support for default framework libraries, fixes #11
closed 2023-02-03 by tsandmann