tsandmann/platform-teensy #12: When building in debug mode, code does not run

krav4, 2023-02-03:
When built in platformio debug mode (build_type) the code does not run on target and teensy becomes unresponsive to tycmd. 

The working platformio.ini:

```
[env:teensy41]
platform = https://github.com/tsandmann/platform-teensy.git#bugfixes/fw_libs
board = teensy41
framework = arduino
lib_deps = 
  SPI
  https://github.com/tsandmann/freertos-teensy.git 
  https://github.com/ftrias/TeensyDebug.git
extra_scripts=pre:c_cpp_flags.py
```

The breaking platformio.ini:

```
[env:teensy41]
platform = https://github.com/tsandmann/platform-teensy.git#bugfixes/fw_libs
board = teensy41
framework = arduino
build_type = debug
lib_deps = 
  SPI
  https://github.com/tsandmann/freertos-teensy.git 
  https://github.com/ftrias/TeensyDebug.git
extra_scripts=pre:c_cpp_flags.py
```

c_cpp_flags.py:

```python
Import("env")

cc = env["CCFLAGS"]
cc += ["-Wno-unused-parameter", "-Wformat=1", "-DUSB_SERIAL" , "-DTEENSY_OPT_FASTER_LTO"]
print(f"CC flags: {cc}")

cxx = env["CXXFLAGS"]
cxx += ["-Wno-volatile"]
print(f"CXXFLAGS: {cxx}")

env.Replace(CXXFLAGS=cxx)
env.Replace(CCFLAGS=cc)
```

tsandmann, 2023-02-04:
I never used or tested `build_type = debug`. 
Are you sure you want to use `-DTEENSY_OPT_FASTER_LTO` with a debug build? That doesn't sound reasonable to me.

krav4, 2023-02-06:
Thanks for your reply. My goal is to be able to use the TeensyDebug while running FreeRTOS. The debug build is just one of the build options specified in the documentation for TeensyDebug. However, if I build without that option and try to #include and use the TeensyDebug, it has the same behavior (code does not run, bootloader inaccessible).

I understand that the TeensyDebug compatibility is not a requirement for this project, but I feel like having the ability to build with platformio "debug" options should be a feature. Removing the TEENSY_OPT_FASTER_LTO option does not make a difference, unfortunately.

If this is not a desired feature for this project, feel free to close the issue. However, I wanted to ask your advice on whether the TeensyDebug would even work with FreeRTOS, and if you ever had any luck with it? https://github.com/ftrias/TeensyDebug

tsandmann, 2023-02-06:
Sure, I'll have a look into `build_type = debug`, it should build working code. But I doubt that it is useful together with an optimized LTO build (in general, not specific to teensy or microcontrollers).

I haven't tested TeensyDebug until now, so I don't know, if it works together with FreeRTOS.

tsandmann, 2023-02-09:
I've pushed an update for the core library which make a `build_type = debug` build work for me at least for the blink example. Maybe you could test that with your project?
But I'm not sure if that changes anything regarding TeensyDebug.

krav4, 2023-02-22:
Hey! sorry was away for vacation.
I don't see any commits later than 3 weeks ago - do you think you could check which branch it was pushed to?

tsandmann, 2023-02-23:
I pushed it to the [default branch](https://github.com/tsandmann/teensy-cores/tree/ts) so it can be tested without a special platformio config.

krav4, 2023-03-08:
> I pushed it to the [default branch](https://github.com/tsandmann/teensy-cores/tree/ts) so it can be tested without a special platformio config.

Hmm it does not want to update the repo for me...
I also had a quick unrelated question - is it possible to change the FreeRTOS memory management pattern? In regular CMake builds I would just compile the correct heap_X.c file, but not sure how to do this with this build system.

[image]

krav4, 2023-03-28:
> Remove the `#43c4b71880bca7798e41087cd49ecd89d759caf2` part from the platform reference.
> 
> > I also had a quick unrelated question - is it possible to change the FreeRTOS memory management pattern? In regular CMake builds I would just compile the correct heap_X.c file, but not sure how to do this with this build system.
> 
> This port uses `malloc()` and `free()` from the C library for memory management, because the teensy framework provides a C runtime that includes `malloc()` and `free()`. Therefore it is not intended to use other memory management patterns.

Thanks!
It works with the debug flag now, thank you.

closed 2023-03-28 by krav4