tsandmann/platform-teensy #13: 'RAM' overflowed during compilation

lekoook, 2023-03-10:
Using commit b92da4fa983a06230c46815f89b6ad788cdde023 causes the following error messages during compilation:

```
/home/lekok/.platformio/packages/toolchain-arm-cortexm-linux/bin/../lib/gcc/arm-cortexm7f-eabi/12.1.0/../../../../arm-cortexm7f-eabi/bin/ld: .pio/build/teensy41/firmware.elf section `.ARM.exidx' will not fit in region `RAM'
/home/lekok/.platformio/packages/toolchain-arm-cortexm-linux/bin/../lib/gcc/arm-cortexm7f-eabi/12.1.0/../../../../arm-cortexm7f-eabi/bin/ld: region `RAM' overflowed by 1152 bytes
```

However, using commit 63670c92387434eca6a8fdd0838637895ef79937 works just fine.

lekoook, 2023-03-11:
You can recreate this failure by creating a new PlatformIO project using Teensy 4.1 as the board. 

In the platformio.ini file, use this:

```
[env:teensy41]
platform = https://github.com/tsandmann/platform-teensy.git
board = teensy41
framework = arduino
```

Then once PIO has configured the project, build it and this error will be reproduced.

tsandmann, 2023-03-11:
That works for me without any errors. 
I'm using this platform for several Teensy 4.1 projects and none of them have this error.

lekoook, 2023-03-11:
Previously it compiled without error for me as well, until I did a "Clean All" in my PIO project.

lekoook, 2023-03-11:
Strangely, manually deleting the `toolchain-arm-cortexm-linux` the package cleared the error.

For anyone facing the same issue, try deleting `.platformio/packages/toolchain-arm-cortexm-linux` folder manual and do a  "Clean All" in PIO. Wait for project to be reconfigured and build again. This worked for me.

closed 2023-03-11 by lekoook