tsandmann/platform-teensy #14: millis() no longer works

krav4, 2023-03-28:
On teensy 4.1:

millis() arduino function returns systick_millis_count, which always stays at 1300 for my project.
I am suspecting this is because the SysTick_Handler in the FreeRTOS port highjacks the systick_isr function address (see below).

Looks like systick_isr is what is supposed to increment it (see EventResponder.cpp), lines 343-351. There is also a comment that this does not work for IMXRT?

`// TODO: this doesn't work for IMXRT - no longer using predefined names
extern "C" volatile uint32_t systick_millis_count;
extern "C" volatile uint32_t systick_cycle_count;
extern "C" uint32_t systick_safe_read; // micros() synchronization
extern "C" void systick_isr(void)
{
	systick_cycle_count = ARM_DWT_CYCCNT;
	systick_millis_count++;
}
`

Should we simply add incrementing the systick_millis_count to the FreeRTOS SysTick_Handler?

krav4, 2023-03-28:
Doing this in port.c did not work:

```C
/*-----------------------------------------------------------*/
#include "EventResponder.h"
#include "Arduino.h"
void xPortSysTickHandler( void )
{
    /* The SysTick runs at the lowest interrupt priority, so when this interrupt
     * executes all interrupts must be unmasked.  There is therefore no need to
     * save and then restore the interrupt mask value as its value is already
     * known. */
    portDISABLE_INTERRUPTS();
    {
        /* Increment the RTOS tick. */
        if( xTaskIncrementTick() != pdFALSE )
        {
            /* A context switch is required.  Context switching is performed in
             * the PendSV interrupt.  Pend the PendSV interrupt. */
            portNVIC_INT_CTRL_REG = portNVIC_PENDSVSET_BIT;
        }
        systick_millis_count++;
        
    }
    portENABLE_INTERRUPTS();
}
```

krav4, 2023-03-28:
**UPDATE:**

The millis() DOES work with the Systick_Handler as it is, so long as the scheduler is started. Before scheduler is started nothing is incrementing the systick_millis_count

tsandmann, 2023-03-31:
> The millis() DOES work with the Systick_Handler as it is, so long as the scheduler is started. Before scheduler is started nothing is incrementing the systick_millis_count.

Does the blink example [example/blink](https://github.com/tsandmann/freertos-teensy/tree/master/example/blink) work for you? It uses `delay()` before the scheduler is started (`delay()` is also using `systick_millis_count`) and works for me as expected.

> Is there anything we can do to make the systick_millis_count increment before the scheduler is started? I am assuming that would require first registering the arduino systick_isr function and then re-registering the Systick_Handler for freertos when we start the scheduler

This is exactly how the current implementation is done.

Do you have a minimal example that causes the same problem?

krav4, 2023-04-01:
Hey Timo - thank you for your reply. That is good news and your implementation makes sense - let me try to put together an example that recreates the problem.

In short, the issue for me was being caused from I2C.begin() being called before the start of the scheduler. It internally has a loop that uses elapsedMillis class to count 16 milliseconds while waiting in idle. Those elapsedMillis stayed at 0 because millis() were not incrementing (systick_millis_count stayed at 1300). Therefore, it was an infinite loop.

tsandmann, 2023-04-01:
Do you have a code snippet that's causing the issue? 
Without reproducing the error, it's hard to figure out what's going on.

krav4, 2023-04-01:
Yes - i am working on getting the error reproduced without sharing company IP

krav4, 2023-04-01:
Okay, I am now able to reproduce it.

My main.cpp:
```cpp
/*
 * This file is part of the FreeRTOS port to Teensy boards.
 * Copyright (c) 2020 Timo Sandmann
 *
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
 *
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 * Lesser General Public License for more details.
 *
 * You should have received a copy of the GNU Lesser General Public
 * License along with this library. If not, see <http://www.gnu.org/licenses/>.
 */

/**
 * @file    main.cpp
 * @brief   FreeRTOS example for Teensy boards
 * @author  Timo Sandmann
 * @date    17.05.2020
 */

#include "arduino_freertos.h"
#include "avr/pgmspace.h"
#include "wire_handler.h"
static void task1(void*) {
    while (true) {
        ::digitalWriteFast(arduino::LED_BUILTIN, arduino::LOW);
        ::vTaskDelay(pdMS_TO_TICKS(500));

        ::digitalWriteFast(arduino::LED_BUILTIN, arduino::HIGH);
        ::vTaskDelay(pdMS_TO_TICKS(500));
    }
}

static void task2(void*) {
    while (true) {
        ::Serial.println("TICK");
        ::vTaskDelay(pdMS_TO_TICKS(1'000));

        ::Serial.println("TOCK");
        ::vTaskDelay(pdMS_TO_TICKS(1'000));
    }
}

FLASHMEM __attribute__((noinline)) void setup() {
    ::Serial.begin(115'200);
    ::pinMode(arduino::LED_BUILTIN, arduino::OUTPUT);
    ::digitalWriteFast(arduino::LED_BUILTIN, arduino::HIGH);

    ::delay(5'000);
    
    if (CrashReport) {
        ::Serial.print(CrashReport);
        ::Serial.println();
        ::Serial.flush();
    }

    ::Serial.println(PSTR("\r\nBooting FreeRTOS kernel " tskKERNEL_VERSION_NUMBER ". Built by gcc " __VERSION__ " (newlib " _NEWLIB_VERSION ") on " __DATE__ ". ***\r\n"));

    ::xTaskCreate(task1, "task1", 128, nullptr, 2, nullptr);
    ::xTaskCreate(task2, "task2", 128, nullptr, 2, nullptr);

    ::Serial.println("setup(): starting scheduler...");
    ::Serial.flush();
    ::Serial.println("Calling Wire.begin()");

    // declaring as static or putting it on the stack produces same issue
    // interestingly, none of the prints work either. Maybe compiler recognizes infinite loop
    // and gives up on running anything at all?
    static WireHandler wire_handler;
    wire_handler.do_wire_stuff();
    ::vTaskStartScheduler();
}

void loop() {}

```

wire_handler.h:
```cpp
#include "Wire.h"
#include "WireIMXRT.h"

struct WireHandler
{
    WireHandler()
    {

    };

    do_wire_stuff()
    {
        Wire.begin();
        ::Serial.println("Calling Wire.beginTransmission()");
        Wire.beginTransmission(0b00001100);
        ::Serial.println("Calling Wire.write()");
        Wire.write(0x3F);
        Wire.write(0x00);
        Wire.write(0x00);   
        ::Serial.println("Calling Wire.endTransmission()");
        Wire.endTransmission(); // hangs here
    }
};
```

platformio.ini:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:teensy41]
platform = https://github.com/tsandmann/platform-teensy.git
board = teensy41
framework = arduino
lib_deps = https://github.com/tsandmann/freertos-teensy.git
build_flags = -Wformat=1 -DUSB_SERIAL -DTEENSY_OPT_FASTER_LTO
upload_flags = -v
upload_protocol = teensy-cli

```

tsandmann, 2023-04-01:
Did a quick test, the program crashes and reboots at the end of `do_wire_stuff()` with "(UNDEFINSTR) Undefined instruction".

Your `WireHandler::do_wire_stuff()` method is missing a return type. If I change it to `void do_wire_stuff()`, the program works as expected  for me (but I did not check the output on the I2C bus).

tsandmann, 2023-04-03:
Ah, I had overlooked that before: you have to move `static WireHandler wire_handler;` prior to any FreeRTOS API call, so before the first `xTaskCreate()` in this case, (or use it in a task).

This behavior is intended by FreeRTOS, see https://github.com/FreeRTOS/FreeRTOS-Kernel/issues/254. If you use FreeRTOS APIs before the scheduler is started, then interrupts are disabled until `vTaskStartScheduler()` is executed. So `systick_millis_count` is no longer incremented until `vTaskStartScheduler()` is executed.

closed 2023-04-03 by tsandmann