tsandmann/platform-teensy #16: .pio\libdeps\teensy41_dbgx_exram\freertos-teensy\src\portable\teensy_4.cpp:186:9: error: 'HardwareSerialIMXRT' not defined

Stevenlawrencehoriba, 2024-07-30:
Updates to 159 are now causign the following compilation errors
.pio\libdeps\teensy41_dbgx_exram\freertos-teensy\src\portable\teensy_4.cpp: In function 'void freertos::yield()':
.pio\libdeps\teensy41_dbgx_exram\freertos-teensy\src\portable\teensy_4.cpp:186:9: error: 'HardwareSerialIMXRT' has not been declared
  186 |         HardwareSerialIMXRT::processSerialEventsList();

Unknown object or librart for HardwareSerialIMXRT, no idea where that is

around thuis area, there is an option to diable HARDWARE SERIAL , what ever that realy means,
I just hope it does not disable all the Serial ports, Testign in progress. With this new falg I can contiue to compile, but havign issues with the std serial port on th eteensy itself , wit hthis new release

PIO settings are..

platform = https://github.com/tsandmann/platform-teensy.git
board = teensy41
framework = arduino
lib_deps = 
	https://github.com/tsandmann/freertos-teensy.git
	https://github.com/PaulStoffregen/SPI
	https://github.com/greiman/SdFat
	https://github.com/nickgammon/Regexp
	ssilverman/QNEthernet@^0.25.0
build_Unflags = -std=gnu++17	
build_flags = -std=gnu++20 -Wformat=1 -fno-strict-aliasing -Wno-deprecated -Wno-unused-variable -Wno-unused-parameter
	-DWAIT_SERIAL -DSERIAL_TESTS -DDEBUG_IFC -DDIAG_TASKS -D QSPI_RAM -DTEENSYDMX_USE_INTERVAL_TIMER 
	-DTEENSY41_RTOS -DARDUINO_TEENSY41 -DUSB_DUAL_SERIAL -DTEENSY_OPT_FASTER_LTO 
	-DLWIP_MDNS_RESPONDER=0 -DLWIP_STATS=0
	-DCOM3_SUPPORT
monitor_speed = 115200
monitor_rts = 0
# remove to stop auto doxygen generation
extra_scripts = post:extra_script.py
upload_flags = -v
upload_protocol = teensy-cli

tsandmann, 2024-07-30:
Looks like you're using an outdated version of the teensy cores library.
Could you run `pio pkg update` in your project folder to update it and test again?

Stevenlawrencehoriba, 2024-07-31:
that has indeed fixed the compilation issues, but with this new release , the USB serial debug port refuses to work, and looks liek it keeps bricking the  teensy. I can get it to re-program with the download USB port (boot USB connection), and I can recover and re-test with the blink program, but building with the above settings, just now bricks the Teensy, and stops the Serial Debug USB port from working. I have removed all the calls to the scheduler, and just a simple main to print something, BUT with the libs and new tool sets built it just bricks the teensy? any ideas. I have tried going backwards, but can not go back as the tool chain is no longer registered on the sites.. so a bit stuck

Stevenlawrencehoriba, 2024-08-01:
further to comments above I went back to the Arduino IDE and just tried a normal program, like the memory test program and that now has a load of connection issues trying to connect serial port, It programs the unit. BUT Then struggles to connect the serial device. this program also loops on WHILE (!SERIAL) , which maybe causing some issues on this board now,.!

tsandmann, 2024-08-01:
It don't think that issue is related to the fork of the teensy cores library, as it uses the same usb serial driver code as the original library.

Which tool chain are you referring to?

Stevenlawrencehoriba, 2024-08-02:
removing the While *(!Serial) makes the system at least detect the serial port now, I can only think its delaying the overall system for some reason, but our RTOS is not starting, It was working a couple of days ago, but that was using older tool chain and framework and more importantly "158".. of teensy.. and without the serial output it is almost impossible to investigate. The new updates have caused more than one issue by the look of it!. Is there any way of going back to the last release V10 and teensy 158 as that used to work ok. I will continue to strip back out system to see if I can find the issue. The blink program you provide sort of works, but the serial is not always reliable as it used to be, and i have tried two teensy' s. There seems to have been some changes around the Serial code in 159 and the core.

tsandmann, 2024-08-03:
If you want to use this platform fork with version 158 of the teensy cores library, you can adjust the following parts of the file `platform.json` in `$HOME/.platformio/platforms/teensy-ts/`:
```json
    "framework-arduinoteensy": {
      "type": "framework",
      "optional": false,
      "owner": "platformio",
      "version": "~1.158.0"
    },
    "framework-arduinoteensy-ts": {
      "type": "framework",
      "optional": false,
      "owner": "tsandmann",
      "version": "https://github.com/tsandmann/teensy-cores.git#1.58.0_ts"
    },
```

Stevenlawrencehoriba, 2024-08-05:
ok thanks. i have been stripping back things, that were working on the older version, and managed to compile the code up. the tasks are compiled in but not started by removing the task creates.. NOW.. All i have running is two tasks our monitor task and some serial test task, the serial task fails to run, BUT from the monitor I see the following diagnostics. I managed to get he serial output sort of working, BUT the heap! is completely depleted. by just compiling in all the new stuff.. ? what's going on here? because the older version used have way more and nothings changed on our side.

Mon Aug  5 13:15:30 2024 UTC | DEBUG   | src\Utilities\System\TaskManager.cpp | diagnosticsTaskInfo:500 | [1]Monitor    P[1], STACK[00000324] STATUS[0]
Mon Aug  5 13:15:30 2024 UTC | DEBUG   | src\Utilities\System\TaskManager.cpp | diagnosticsTaskInfo:500 | [2]Serials    P[7], STACK[00000834] STATUS[2]
Mon Aug  5 13:15:30 2024 UTC | DEBUG   | src\Utilities\System\TaskManager.cpp | diagnosticsTaskInfo:513 | Free Heap Size = 52
Mon Aug  5 13:15:30 2024 UTC | TRACE   | src\Tasks\Diagnostics\Monitor.cpp | monitor_task:79 | Monitor Start

tsandmann, 2024-08-10:
See tsandmann/freertos-teensy#29

closed 2024-08-10 by tsandmann